christianhaitian/arkos #1: Several issues

jedinat, 2020-11-15:
There is no fbneo folder. Adding the folder doesn’t add the games.

Putting my MAME-2003 games in the MAME folder results in them not running. It won’t let me change the default core (no core even appears in the setting). Putting them in the MAME-2003 folder works, but there’s no artwork for it in the menu, so it’s just some barely visible black text.

The brightness setting automatically jumps down a notch or two when you implement it. So if you put it at 1% it drops to zero and the screen turns off.

I can’t find any manual way to update the game list. (does it update when restarting only?)

christianhaitian, 2020-11-16:
There is no fbneo folder because fbneo based games should be loaded in the arcade folder per the information posted in this [wiki link](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#arcade).  

The Mame-2003 folder artwork issue has been resolved in an update released today.  Just scroll to the Options menu just after scummvm and click on update.  Make sure you have a wireless dongle connected and configured.

The brightness setting automatically jumping down a notch when set from the menu is an issue with emulationstation-fcamod which this distro uses.  Couple of workarounds for the issue:

- Use the global hotkey instead to adjust the brightness.  Information about global hotkeys for the rg351p are available in the wiki in the faq section [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-arkos).

- If you do want to set the brightness from the menu, set it 1 percent higher than your actual value before exiting the option.

The game list is updated automatically upon restarting.  If you're looking to add more information about your games, you can set those per game by highlighting a game and hitting select then scroll down to **Edit this game's metadata** for the game.

jedinat, 2020-11-16:
Thanks for the great info!

closed 2020-11-16 by jedinat

jonathan17819, 2025-02-21:
first issue