christianhaitian/arkos #2: Dreamcast will not boot .7z roms

metaColin, 2020-11-17:
#### Issue description
7zipped Dreamcast roms will not boot.

#### Steps to reproduce the issue
1.  Copy .7z formatted Dreamcast rom to ROMs folder, ensure you have the needed bios installed.
2. Try to launch it from the list of Dreamcast games.

#### What's the expected result?
Game should load.

#### What's the actual result?
Screen attempts to load title and then returns to the previous menu.

#### Other notes
The same roms will boot if I unzip them (there are cue and bin files in each 7z). The entry on the ArkOS wiki says Dreamcast supports .7z files, so I'd really like to take advantage of the space-saving instead of expanding my entire romset to occupy more space.

I also tried manually opening Retroarch or Retroarch32 and trying to manually launch with each available Dreamcast core, but none of them would load the roms. (I tried with several different roms to be certain).

metaColin, 2020-11-17:
I can open a separate issue but it seems like PS1 titles have the same issue with zipped roms (cue/bin).

christianhaitian, 2020-11-17:
I can confirm that there's an issue with .7z and the flycast and other dreamcast core.  I tried various .7z sources and the result is the same load content error.  I suspect the issue needs to be resolved in the libretro code for this but I'm not sure if this is an issue that just impacts arm based systems or not.  I'm hoping a future update will address that.  I've updated the wiki to reflect this issue so others will know.  Thanks for reporting this.  I was able to boot a .zip ps1 title that contained just a bin without issue.

metaColin, 2020-11-18:
It seems like the .7z issue with ps1 titles is very inconsistent, I would say 1 in 3 roms I've tried does eventually load.

It looks like other libretro users have encountered the same trouble with zipped ps1 assets on various other platforms. I think it just pushes the RAM/CPU beyond what it can handle depending on the payload to be decompressed.

Seems like the best workaround for keeping a small filesize is to convert all my ps1 roms to pbp. It's a good solution, but it's gonna take a while :(

metaColin, 2020-11-18:
As for Dreamcast, since you were able to reproduce the issue with 7z roms, I guess I'll have to convert that set to CHD.

christianhaitian, 2020-11-19:
Closing as the issue with flycast and .7z is not something that is fixable here.  Has to be fixed by libretro.

closed 2020-11-19 by christianhaitian