christianhaitian/arkos #5: Gambette roms crashing

ParkRomn116, 2020-11-19:
I updated to the ArkOS and have a few different played gbc games and have come across the games crashing ( screen turns white and stays for a while until I hard reset the device. 
This has happened to Pokémon Crystal And Gold so far. 
In these games I have set up hot keys, I have used fast forward a couple times. I never used the power button and always turned off the device from the menu. 
Hope this helps 
Thanks  so much for your effort, I'm fairly new to these things and I have enjoyed my device so much bc of this update.

ToniBC, 2020-11-19:
I have tried them and those games do not fail, they start without problems. So it has to be another problem or something you touched on from the emulator settings. At least I couldn't reproduce that error.

PS: It's Gambatte xD

christianhaitian, 2020-11-19:
Also note that there are other gameboy emulators you can try if the default gambatte one doesn’t work for a particular game.  See this [link](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-change-emulator-cores-in-arkos) to the FAQ in the wiki for more details on how to change the default emulators.  List of available emulators per system are available in this [link](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information)

closed 2020-11-19 by christianhaitian