christianhaitian/arkos #6: Separate Saves from games

ToniBC, 2020-11-19:
Retroarch has several options to separate the games by core into folders, all of which are usually located in the same folder with the name saves.

Here I see that these saves are all in the same folder of the rom, mixing everything and I think it would be better to separate these things.

That is, you would have to create a saves folder and within, by default, activate the storage of them separated by cores, everything would be more organized.

Thank you.

christianhaitian, 2020-11-24:
I've reviewed the possibility of doing and according to the retroarch documentation, a game, core, or content directory override would be needed for each emulator to allow this functionality.  This would further complicate the management and support of this distribution.  It's best that the end user handle that need on their own if that's their preference.

closed 2020-11-24 by christianhaitian

ToniBC, 2020-11-25:
It does not generate any problem, I in the share folder where we have all the roms and bios, create a folder with the name of "saves" and in Retroarch it indicates as in the normal versions of Retroarch, that this is the folder, after that Separate by core or I think you can indicate that you separate them by folders and voila, all in one place without spoiling the roms directories.

I think this way is much better and neater, it's how Retroarch originally works. But hey, they are tastes, I always use it like that, others may like that everything is mixed.