christianhaitian/arkos #7: Scraping Platforms Problems

ToniBC, 2020-11-19:
At the moment, the Sega Master System platform does not work by default for Scrapper, because it identifies it as "sms" instead of "mastersystem".

At the moment it is the one I found, that I repaired it manually, but it would be fine for new users.

If I see other things I will add them.

Thank you.

christianhaitian, 2020-11-19:
How did you repair this?

ToniBC, 2020-11-19:
In /etc/emulationstation/es_systems.cfg

		<name>mastersystem</name>
		<fullname>Sega - Master System</fullname>
		<path>/roms/mastersystem/</path>
		<extension>.mdx .MDX .md .MD .smd .SMD .gen .GEN .bin .BIN .cue .CUE .iso .ISO .sms .SMS .gg .GG .sg .SG .68k .68K .chd .CHD .zip .ZIP .7z .7Z</extension>
		<command>sudo perfmax; /usr/local/bin/%EMULATOR% -L /home/ark/.config/%EMULATOR%/cores/%CORE%_libretro.so %ROM%; sudo perfnorm</command>
		   <emulators>
		      <emulator name="retroarch">
		 	<cores>
		 	  <core>genesis_plus_gx</core>
		 	  <core>picodrive</core>
		 	</cores>
		      </emulator>
		      <emulator name="retroarch32">
		 	<cores>
		 	  <core>picodrive</core>
		 	</cores>
		      </emulator>
		   </emulators>
		<platform>mastersystem</platform>
		<theme>mastersystem</theme>
	</system>

The line -platform>**sms**</platform> for this other -platform>**mastersystem**</platform>

The line of the platform almost at the end, which does not let me put the text right here. With this it detects the platform.

christianhaitian, 2020-11-22:
This has been included as a fix in the last update.  Thank your for reporting and providing the solution.

closed 2020-11-22 by christianhaitian