christianhaitian/arkos #13: ArkOS

SGPork, 2020-11-21:
I have only one issue with ArkOS. It does not have the option to switch off the WiFi. My WiFi is always on despite not connected on my RG351P when I updated to ArkOS.

A future update of ArkOS with an option to switch on/off WiFi would be highly appreciated.

superxor, 2020-11-23:
@christianhaitian sudo nmcli radio wiki off, would be simple solution to put into the options menu

SGPork, 2020-11-24:
Is the enhancement possible? Any estimated timeline for this option to be updated in ArkOS?

christianhaitian, 2020-11-24:
Is this a request because you have a rare unit with an internal wifi chip?  If not, why not simply unplug the wireless dongle when not in use?

SGPork, 2020-11-25:
This is definitely a request. My WiFi is built-in, there is no dongle to unplug. The stock OS Emuelec has an option to switch on and off WiFi but ArkOS doesn’t. So an update on ArkOS to switch on and off the WiFi is definitely appreciated.

christianhaitian, 2020-11-25:
This has been accommodated with today's update.  The ability to disable and enable wifi will be available under the Options/Advanced section after completing an update in the Options section.

closed 2020-11-25 by christianhaitian

SGPork, 2020-11-25:
Thanks! You are the best!