christianhaitian/arkos #14: Hotkey Screen Brightness

Chipptune, 2020-11-21:
Is there a possibility of creating a hotkey or button pressed to control brightness levels? Would be nice to adjust on the fly. 

Thank you for your hard work. 😊

swatgoss, 2020-11-21:
Hi,

Isn't the MAIN MENU displayed in [this video](https://www.youtube.com/watch?v=YBYMKrf775o&t=310) (around 5min) able to answer your need ?
(never touched a RG351P / arkOS as of now, but if this menu is already accessible while in a game, this would also be able to allow more hardware control in the like of what you ask)

christianhaitian, 2020-11-21:
This is already possible by holding the r3 button and pressing up or down on the gamepad.  See this link for global hotkeys.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS

closed 2020-11-21 by christianhaitian