christianhaitian/arkos #16: RetroArch crashes when loading remap file in FBNeo

retrogamecorps, 2020-11-21:
When playing arcade games in TATE (vertical) mode, I need to go in and adjust the Port 1 direction controls so that up is right, and so on.  Once direction the controls are adjusted, I typically will save a game remap file (within the Controls menu), and then when loading a different game that requires vertical controls, I can just load that other remap file, and then immediately save it for the game I'm playing too.  It creates a new .rmp file for my game without having to actually readjust all of the controls by hand.  This method works well on stock and 351ELEC OSs, as well as with the MAME 2003-Plus core in 64-bit RetroArch on ArkOS.

But when using the FBNeo core in 64-bit RetroArch in ArkOS, when I try to load an .rmp file from the Controls menu (Quick Menu > Controls > Load Remap File), RetroArch crashes.  I tried looking online to see if this is a known issue with RA, but didn't see anything.

You can recreate this issue by saving a game remap file for one game, then trying to load that game remap file for another game.  Thanks for doing what you do!

christianhaitian, 2020-11-21:
Building and testing an updated version of this core to see if the issue is resolved.

christianhaitian, 2020-11-21:
Accidentally closed this issue.  Reopening

christianhaitian, 2020-11-22:
There's a new retroarch executable available to resolve this issue.  Just update from the options menu.

closed 2020-11-22 by christianhaitian