christianhaitian/arkos #17: Feature request: Steam Link port

retrogamecorps, 2020-11-21:
If possible, I would love to see a port of the Raspberry Pi version Steam Link app for ArkOS, similar to how the Raspberry Pi version of Pico-8 can be loaded as a port.  This would allow users to stream their Steam library (from Mac or Windows) to the device.  In addition to PC games, Steam now allows for any app/game to be imported into your library, so you could essentially stream standalone emulators like Dolphin, Yuzu, or Cemu to the device, or frontend systems like OpenEmu, LaunchBox/BigBox, or even the desktop version of RetroArch.

https://steamcommunity.com/app/353380/discussions/6/2806204039992195182/

Kernotmdl, 2020-11-21:
Maybe not steamlink, but Moonlight streaming is already ported to ubuntu-odroid ho advance, same chip, I tried to install It with t'he instructions from the developer on this link without luck.
But muy Linux knowdlege is someway outdated...
https://forum.odroid.com/viewtopic.php?t=37401
Have It on Arkos would be so sweet.

magnific0, 2020-12-08:
I don't think steamlink will be straight forward. I tried but leaving it for now as the software seems pretty Raspberry specific and I'm afraid of messing up ArkOS. 

I bypassed the debian packages for now, which just seems to take care of the dependencies and has a wrapper that pulls in the latest software on launch.

Login as ark user
```
cd ~
wget http://media.steampowered.com/steamlink/rpi/steamlink-rpi3-1.1.65.164.tar.gz
tar -xvf steamlink-rpi3-1.1.65.164.tar.gz
touch ~/steamlink/.ignore_cpuinfo
ark@rg351p:~$ /home/ark/steamlink/steamlink.sh
```
The empty file is there to bypass a system check for an Raspberry.

Running `steamlink.sh` gives a couple of errors. Searching for `cec-client` and `libbcm_host.so` seem pretty Raspberry specific. cec-client does HDMI switching? BCM is for Broadcom chips in the Raspberries.

```
/home/ark/steamlink/steamlink.sh: line 132: cec-client: command not found
ioctl( UI_DEV_SETUP ) failed: Invalid argument
shell: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
screenblank: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
```

christianhaitian, 2021-01-03:
In reviewing the steamlink info, I won’t be able to bring that to ArkOS.  I may look to include moonlight-qt in the future as that is more feasible.

closed 2021-01-03 by christianhaitian

retrogamecorps, 2021-01-03:
Okay, thanks for the update.  I figured it might not be possible.