christianhaitian/arkos #19: Noise after updating to version 1.2

Sergiomg24, 2020-11-22:
I  have version 351 without wifi but when I update arkos to version 1.2 I notice a sound even if it is set to the minimum. Before updating I did not notice that noise, is it possible that updating the kernel affects 3.51 versions that came without wifi? thanks for your excellent work

christianhaitian, 2020-11-22:
My unit came with no internal wifi and I have not noticed this issue.  I loaded a previous version without the new kernel and don't notice a sound there either.  Can you try the stock OS and see if the sound happens there as well and report?

Sergiomg24, 2020-11-22:
The same thing happens with the operating system that comes from the factory. 

The noise is not annoying, if I do not bring my ear to the console it is not noticeable. I noticed it because when I upgraded to arkos 1.2 (using the wifi adapter, I hadn't used it before) there were noises while arkos was upgrading. then I paid attention. It does not matter if the volume is at minimum or maximum, it sounds the same, it seems something foreign to games As I said it is a noise that is not very annoying and if I do not bring my ear close to it, it is hardly noticeable, but I do not know if it will have to see the use of the wifi adapter or if it was heard before and I had not noticed

jswagner, 2020-11-22:
It's a known issue with the built-in WiFi and is why Anbernic removed the WiFi module from the system.  Fix: https://www.youtube.com/watch?v=pQYeQW8bwqQ

christianhaitian, 2020-11-22:
Since the issue happens with the stock card I would say it’s hardware, however, since you indicated that you only notice it when you put the unit up to your ear, just don’t do that.

closed 2020-11-22 by christianhaitian