christianhaitian/arkos #20: USB Wifi dongle: Works (emuelec) | Doesn’t work (ArkOS)

Kratos291, 2020-11-22:
So my USB wifi dongle that perfectly works with the stock 3.7 emuelec firmware as well as 351elec, doesn’t work with ArkOS. The activity LED doesn’t even light up. Tried everything: Restarting, Reflashing, different USB OTG adapters... nothing works. I changed back to the stock firmware and voila, the dongle works there. Please fix. 

My dongle: https://www.amazon.com/-/de/dp/B07PB1X4CN/

christianhaitian, 2020-11-22:
This issue was resolved with the latest 1.2 image available from the wiki as of yesterday.  Please download the new image and try your usb wifi with that.  I will leave this issue open for now in case you still have issues with your wifi stick.

Kratos291, 2020-11-23:
Appreciate the quick response but still doesn’t work.

christianhaitian, 2020-11-23:
Did you follow these instructions from the wiki to configure your wifi?  
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-configure-my-wifi-connection-in-arkos

Kratos291, 2020-11-23:
Yes, these instructions expect a detected wifi dongle, but my dongle isn’t detected it seems because the LED light that is usually on and blinking is off all the time. With emuelec and 351elec, the LED of my dongle lights up a second or so after I press the power button to turn the device on. With ArkOS it doesn’t light up at all.

superxor, 2020-11-23:
There's no driver in the linux ubuntu kernel for your dongle included. You'd have to manually build and load it

sudo apt install git dkms
git clone https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
sudo ./dkms-install.sh
sudo modprobe 88XXau

superxor, 2020-11-23:
@christianhaitian just FYI regards faq #what-wifi-adapters-work, the usb dongle that Anbernic sends packaged with their devices also works.

Kratos291, 2020-11-23:
I will try to build it myself and report back. Thanks.

My 351p I bought from a european retailer didn’t include a wifi dongle in the package.

superxor, 2020-11-23:
I bought directly from Anbernic via Aliexpress and they shop them incl 64GB sd and wifi dongle from their EU warehouse. 
But if you google your dongle and ubuntu you'll see many people having issues and having to build driver themselves. It's just not in mainline linux kernel

Kratos291, 2020-11-23:
Just to make sure, I will only have to build this driver myself once and new updates to ArkOS won’t wipe the driver, right?

christianhaitian, 2020-11-23:
The 8812au driver is included with ArkOS.  This chipset is similar to what's included with the Netgear A6100 which I have and can confirm that the latest ArkOS works fine with.  That's not the issue.
https://pasteboard.co/JBHv5d8.jpg

christianhaitian, 2020-11-23:
What is the name of the ArkOS image you downloaded that you flashed to your SD card?

Kratos291, 2020-11-23:
The file name is ArkOS_RG351P_v1.2.img that I extracted from the ArkOS_RG351P_v1.2.7z from the gdrive link on the bottom of the wiki.

I reflashed again but still not detected/not working. Could it be that the issue is related to the USB OTG adapter? I have five different USB3.0 OTG adapters, I tried them all but they didn't make a difference. I don't know if the USB OTG adapter Anbernic included in the package (mine wasn't shipped with one) is a USB.2.0 OTG adapter and I don't have a USB2.0 adapter at hand to test it myself.

superxor, 2020-11-23:
Only way to know what the issue is would be to see dmesg output from the console.
If you can login via ssh, plugin the dongle, run 'dmesg' and pastebin that output somewhere. The last lines should show if its a) detected at all b) if its detected what it appears as

christianhaitian, 2020-11-23:
> The file name is ArkOS_RG351P_v1.2.img that I extracted from the ArkOS_RG351P_v1.2.7z from the gdrive link on the bottom of the wiki.
>
> I reflashed again but still not detected/not working. Could it [...]

It shouldn't since the same combo works with the stock firmware and 351elec as you stated.

christianhaitian, 2020-11-23:
> Only way to know what the issue is would be to see dmesg output from the console.
> If you can login via ssh, plugin the dongle, run 'dmesg' and pastebin that output somewhere. The last lines should show if its a) detected at all b) if its detected what it appears as

I agree with this.  If you follow these directions (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-can-i-access-a-terminal-physically-on-ArkOS) , you can get a terminal on the unit and you would just need to connect a keyboard to the other usb c port and do a lsmod with the wifi adapter plugged in to see what driver is loaded.

Kratos291, 2020-11-23:
Where is the log saved?

superxor, 2020-11-23:
Easiest to do via ssh and just copy last 10 lines from your terminal

christianhaitian, 2020-11-23:
No logs are saved.  Just use a smartphone camera and snap a pic and post.  or you can do the following to get a file of the output:

lsmod > lsmod.txt

Kratos291, 2020-11-23:
ok, here you go.
lsmod: 
[image]

dmesg: 
[image]

christianhaitian, 2020-11-23:
Hmmm...yea it loads nothing

christianhaitian, 2020-11-23:
Post a pic of the following:

dmesg | grep usb

Kratos291, 2020-11-23:
just fyi, on my second USB port, I have a wireless receiver for my wireless keyboard plugged in.

[image]

superxor, 2020-11-23:
try: sudo modprobe rtl8812au

Kratos291, 2020-11-23:
> 
> 
> try: sudo modprobe rtl8812au

This didn't do anything.

superxor, 2020-11-23:
now check lsmod again

christianhaitian, 2020-11-23:
Wow.  Same serial number as my unit but the rtl8812au driver registers just fine

superxor, 2020-11-23:
@Kratos291 you sure its not working? did u check the wifi menu, that it doesnt show networks? the menu is a bit confusing. Linux recognizes the adapter, it should be fine

christianhaitian, 2020-11-23:
The lsmod pic he posted earlier didn't show the driver loaded though.

christianhaitian, 2020-11-23:
I think the IDVendor and idProduct strings may be what's causing the issue here.

Kratos291, 2020-11-23:
> now check lsmod again

lsmod: 
[image]

dmesg: 
[image]

And yes, I’m sure it didn’t work and still does not work. There is no wifi to set up in the settings and the LED of my dongle still doesn’t light up.

Kratos291, 2020-11-23:
As you can see there is nothing to set up.
[image]

christianhaitian, 2020-11-23:
Did you reboot to go back to emulationstation?

superxor, 2020-11-23:
plug out and plug in wifi aagain and check again please if its working

Kratos291, 2020-11-23:
> 
> 
> Did you reboot to go back to emulationstation? If so, that means the driver didn't load as you had to do it manually.

I did that: sudo systemctl restart emulationstation

> 
> 
> plug out and plug in wifi aagain and check again please if its working

Not working.

superxor, 2020-11-23:
Doesn't make sense, I'd love to get my hands on that thing :)

christianhaitian, 2020-11-23:
Found some interesting information here:  https://wiki.gentoo.org/wiki/AC1200_Wireless_Adapters#Realtek_RTL8812AU
especially this:

The RTL8812AU driver will not install correctly if the USB adapter is plugged-in before the driver is installed.
The RTL8812AU driver may not install correctly unless the kernel configuration enables some of the built-in Realtek wireless drivers (including those in staging) as modules

I'm gonna double check my kernel configurations to verify that the staging modules are included.

christianhaitian, 2020-11-23:
I made the suggested changes and will upload a test image for you to try.  Hopefully it resolves the issue and helps add additional support for more devices.

Kratos291, 2020-11-23:
Ok I'll go to sleep now. I will test the image when I can.

christianhaitian, 2020-11-23:
Thank you.  It should be available shortly.

christianhaitian, 2020-11-23:
Here's the link to the new test image.  Please download it and try it and report if it resolves your issue or not.  Thanks.

https://mega.nz/file/DMoCjbBS#0dANP1lryenK_ldklD90npo6p_TAnJuDLkUzvDpnlAU

Kratos291, 2020-11-23:
I tested the image and it’s still not working. I will try the method where you load the driver manually tomorrow.

Btw, on the boot screen it was still showing ArkOS 1.2 (in case it was supposed to show 1.3).

christianhaitian, 2020-11-24:
I have another image that I'm currently packaging that I believe should resolve the compatibility issue with your existing adapter.  Apparently, the correct driver for your adapter is the 8821 driver, not the 8812 as we previously investigated.  I updated the kernel configuration to add some additional supported adapter which includes the Archer T2U.  The update should be available in a couple of hours.  I will post the link to it hear.

christianhaitian, 2020-11-24:
Please try this new image and report your results.  https://drive.google.com/file/d/1OOLZIyo7M14B8_L0u5gINSw-3dyyhkGp/view?usp=sharing

Kratos291, 2020-11-24:
I can confirm with this new image my Wifi dongle is working now. Thanks so much, awesome work!

PS: Since it's somewhat related, I noticed that with a wifi dongle plugged in, as soon as you plug in anything else in the second USB-C port, the wifi dongle stops working. However, if you have a wifi dongle and anything else plugged in your device while it's turned off, and you turn it on, the wifi dongle works without issue. This however happens with any other firmware for the 351p as well so it might not be ArkOS related but still it would be pretty cool if ArkOS is the only firmware that fixed this bug.

christianhaitian, 2020-11-24:
Great!  And that issue with plugging a second product in (Like power) is known due to the usb c ports missing some noise filtering components.  Enjoy!

closed 2020-11-24 by christianhaitian

NaDne73, 2022-02-10:
hello, can you add this driver to ark os? Realtek RTL8811CU/RTL8821CU tenda u9 dongle