christianhaitian/arkos #21: exFAT partition not listed by Windows 10

paprikania, 2020-11-23:
Hi, I only just realized - when setting up a new TF - card that the exFAT partition is no longer listed in or accessible via the Explorer in Windows 10. When I set up my initial card a week ago, that was possible. Not being able to read or write that partition is quite annoying. Is it possible to adapt the formatting/partitioning process so that it is again accessible via WIndows 10?

superxor, 2020-11-23:
from wiki:
FOR WINDOWS 10 1703 or newer USERS: If you don't see a drive letter named EASYROMS when you plug the SD card into a card reader, it's most likely that Windows did not automatically assign a drive letter to that partition on your SD card. This can be resolved by going to disk management (type disk management in the search bar in Windows 10 and select the first control panel app that comes up at the top as the best match), then going to the SD card with the EASYROMS partition label, then assign a drive letter to the EASYROMS partition by right clicking on the EASYROMS partition and selecting Assign Drive Letter or Change Driver letter and Path, then follow the directions from there. Once completed, the drive should show up under My Computer. You typically only need to ever do this once on the Windows machine.

paprikania, 2020-11-23:
Should RTFM, sorry... but thanks!

closed 2020-11-23 by paprikania