christianhaitian/arkos #22: FBNeo does not load Neo Geo games with last update

mfogue, 2020-11-23:
I updated via WiFi to the latest version and now I can't load the Neo Geo roms using Retroarch, I only get a black screen and it doesn't seem to do anything else. They were working fine before the update.

I tried other roms (CPS1, CPS2...) using the same core (FinalBurn Neo v0.2.97.44 e8106ea65) and they are working as expected, no issues there, only with Neo Geo roms.

christianhaitian, 2020-11-23:
I was able to load 2020 Super Baseball, Art of Fighting, Art of Fighting 2, and Metal Slug without issue on my end after the same updates.  I even tried on a new image and completed the same updates, and same result.  I can't replicate this issue on my end.

superxor, 2020-11-23:
Works for me, tried different games?

mfogue, 2020-11-23:
It seems like a weird issue, I couldn't load any of the games in the neogeo directory, but I deleted the games and loaded them again and now they work. I still don't know what happened, but it doesn't seem to be related to the new version. Not sure if it's worth studying the issue or just close it

christianhaitian, 2020-11-23:
Closing due to not being able to replicate and issue was resolved by reloading the games into the neogeo directory.

closed 2020-11-23 by christianhaitian