christianhaitian/arkos #24: Move music folder to bios folder?

retrogamecorps, 2020-11-24:
Hi, I'm finding that a lot of people love the idea of the background music feature but are put off by having to figure out how to configure SSH for this one thing.  Is it possible to make the default BGM folder something like EASYROMS/bios/music so users can just drag/drop using the ExFAT partition?

christianhaitian, 2020-11-24:
This was already planned for the next update that is releasing in the next couple of hours.  There will be a roms/bgmusic folder created during the update.  Existing music content in the .emulationstation/music folder will be moved to that folder and a symbolic link will be created to the bgmusic folder from .emulationsation/music.

closed 2020-11-24 by christianhaitian

retrogamecorps, 2020-11-24:
Great minds think alike.  Thanks! :)