christianhaitian/arkos #27: Emulators

alvaromago78, 2020-11-24:
Hi Christian

Thanks for it!

I dont see any plattforms like Commodore Amiga, Atari St or Spectrum. I have entered in retroarch and it is installed. I added roms but it is dont appear. Are these systems compatible with Arkos?

Other question, Caprice, amstrad emulator, show reduced screen in games, can i change it? I like other Amstrad emulator called Fuse but I dont how install it.

Thanks in advance.

christianhaitian, 2020-11-24:
Make sure you review this entry in the wiki for available emulators and accepted file extensions and other information:  https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information

alvaromago78, 2020-11-24:
Thank you very much! :)

closed 2020-11-24 by christianhaitian