christianhaitian/arkos #30: Updating to 1.2 Sticks

SirMatlock, 2020-11-25:
Not sure if it’s supposed to, or not. 
When the update gets to removing the Switch Theme, it stops. Have to reset to go back. 
The device shows its updated to 1.2, but I’m not sure how to verify.

Also, do you have a change long for 1.2?

Keep up the great work.

christianhaitian, 2020-11-25:
Rerun the update.  You may have had a network blip that caused the issue.  You should be to version 1.3.  See here for the changelog.  https://raw.githubusercontent.com/christianhaitian/arkos/main/changelogs/rg351p-changelog

christianhaitian, 2020-11-26:
I tested the update again, and what's taking so long is the download of the epicnoir theme.  On a weak signal, the theme takes about 5 minutes to download.  Once completed, the rest of the update goes quick.  I recommend when you run the update again, be sure you've got a strong signal and a fast internet connection or that particular update will take a good while.

closed 2020-11-26 by christianhaitian