christianhaitian/arkos #31: v1.3 Installation Hang

couchp95, 2020-11-26:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
Installation hang with v1.3

### How can the issue be reproduced? (If applicable)
1. Download the image from MEGA.
2. Uncompress the image with 7zip
3. Burn the img to 128GB TF with the USB Image Tool by Alexander Beug
4. Insert into rg351p and boot
5. Hang with the display of "The partition table has been altered." as attached.
[image]

### What device are you using?
rg351p

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
v1.3

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-11-26:
What is the make and model of your SD card?

Also, please uninstall Paragon Linux file system if you have this installed.

couchp95, 2020-11-26:
What is the make and model of your SD card?
ADATA Premier 128GB
https://www.amazon.com/ADATA-Premier-MicroSDHC-Adapter-AUSDX128GUICL10A1-RA1/dp/B077QJ1HVT?ref_=ast_sto_dp
Also, please uninstall Paragon Linux file system if you have this installed.
Uninstalled before burn the TF card

christianhaitian, 2020-11-26:
The issue may be that brand of SD card.  I'm not familiar with how reliable Adata SD cards may be.  Below is a link to the cards that are of known good quality.  Also, you can try to use win32diskimager instead to see if you get better results with that.

https://www.techradar.com/news/best-sd-and-microsd-memory-cards

superxor, 2020-11-26:
How long did you wait? I remember on my 64GB I had to wait 2-3min before it continued because SD cards are slow and its checking the FS iirc.
Give it 10min or so

couchp95, 2020-11-26:
> How long did you wait? I remember on my 64GB I had to wait 2-3min before it continued because SD cards are slow and its checking the FS iirc.
> Give it 10min or so

Long enough. I tried again with Samsung 64GB and it's working properly. Not sure why ADATA doesn't work as it's a reliable brand.

couchp95, 2020-11-26:
Tried another Sandisk 128GB and it's working fine.
Please close this issue. Thanks for your help.

closed 2020-11-26 by couchp95