christianhaitian/arkos #34: Add OpenBor enhancement

myzar, 2020-11-26:
OpenBor is the only emulator missing for the rgp-351 , it's missing in all the distros because neither the dpad nor the analogs work.
The joysource must need some patching to work i guess i've looked in to it but can't figure it out yet

christianhaitian, 2020-11-30:
What source did you review?

myzar, 2020-11-30:
i was looking here https://github.com/DCurrent/openbor/blob/master/engine/sdl/joysticks.c , i've also tried all the binaries for the rk without luck

christianhaitian, 2020-12-15:
Quick Update, OpenBor does run on the unit but the controls are problematic.  I will review this in the near future in hopes of resolving it so it can be included with a future update.

microton79, 2021-01-05:
Hi!

i dont know if it helps but i found a joytokey file for openbor that is used in a different distro (retrofe/coinops). I rename it to txt extension but the ext is amgp. In RetroFE/Coinops openbor runs and works with controller

[Profile.gamecontroller.txt](https://github.com/christianhaitian/arkos/files/5770760/Profile.gamecontroller.txt)

christianhaitian, 2021-01-29:
This has been accommodated with this evening's update.  Just update from the options section.

* for pak loading, see emulators and ports section (https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#openbor)
* To exit the emulator, hit the Start button then select exit in the menu for most paks or hit Select+Start

Enjoy!

closed 2021-01-29 by christianhaitian