christianhaitian/arkos #36: Add Tic-80 support

jlbrd, 2020-11-27:
Please add tic-80 emulator

christianhaitian, 2020-11-28:
While the emulator is available, none of the included themes in Emulationstation for ArkOS support Tic-80 currently.  Will wait until at least nes-box supports it and then will add this emulator.

jlbrd, 2020-11-28:
Thank you for your reply. Should I ask NesBox to add the theme for Emulationstation?

jlbrd, 2020-11-29:
Hello, The TIC80 core is listed at Retroarch. So I launched retroarch 64. Installed TIC80 by downloading the core. Created a directory /home/roms/tic80 and copied tetric.tic. Enbled logs in retroarch. Launch retroarch, load the core, load the tetris.tic content. The screen turns black and then returns to the menu. In the logs:

`[INFO] === Build =======================================
[INFO] Capabilities:  ASIMD
[INFO] Built: Nov 22 2020
[INFO] Version: 1.9.0
[INFO] Git: 175205c1e6
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [CORE]: Loading dynamic libretro core from: "/home/ark/.config/retroarch/cores/tic80_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at /home/ark/.config/retroarch/config/TIC-80/TIC-80.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /home/ark/.config/retroarch/config/TIC-80/tic80.cfg.
[INFO] [Overrides]: No game-specific overrides found at /home/ark/.config/retroarch/config/TIC-80/tetris.cfg.
[INFO] [Environ]: SET_SUPPORT_NO_GAME: no.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Remaps]: remap directory: /home/ark/.config/retroarch/config/remaps
[INFO] Saving files in content directory is set. This overrides other save file directory settings.
[INFO] Saving save states in content directory is set. This overrides other save state file directory settings.
[INFO] [Overrides]: Redirecting save file to "/roms/tic80/tetris.srm".
[INFO] [Overrides]: Redirecting save state to "/roms/tic80/tetris.state".
[INFO] [CONTENT LOAD]: Loading content file: /roms/tic80/tetris.tic.
[INFO] Did not find a valid content patch.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.`

Your help would be welcome

christianhaitian, 2020-12-02:
Tic-80 support has been added with the latest update.  Just go the Options section and select update to receive the latest updates.  You can also review this [changelog](https://github.com/christianhaitian/arkos/raw/main/changelogs/rg351p-changelog) for a list of all the rest of the included updates.

closed 2020-12-02 by christianhaitian

jlbrd, 2020-12-02:
Hello and thank you for your work. The update is installed. Now a Tic80 menu and the installed games are displayed. But I still have the same launch problem as before. I don't know if it's caused by emulationstation, retroarch or ArkOS. And no error in the Retroarch log. Do you have an idea ?

christianhaitian, 2020-12-02:
.tic games launch without issue for me.  I tested mario bros., up dog, and cauliflower power.

jlbrd, 2020-12-02:
I flashed the sdcard and after updating everything works. Thank you for everything.