christianhaitian/arkos #38: Add Amiga Retroarch core support.

neil-9000, 2020-11-28:


neil-9000, 2020-11-28:
Retroarch supports Commodore Amiga via the PUAE core, it would be nice to be able to use this.

christianhaitian, 2020-11-28:
Last I recalled, the performance was not good for this architecture.  Have you tested by going directly into retroarch are downloading the core and launch some games?

neil-9000, 2020-11-28:
Yeah i have tested it, its not 100% ill give you that, but the games i tested were around 50fps at all times, which is what they should be as a PAL system. I tested Robocop, Another World/Out Of This World and Sensible Soccer and all performed at or very close to full speed. Certainly a far better experience than most Dreamcast and PSP games on this device if you ask me.

christianhaitian, 2020-11-30:
PUAE core support has been included with today's update.  Just go the Options section and select update to receive the latest updates.  You can also review this [changelog](https://github.com/christianhaitian/arkos/raw/main/changelogs/rg351p-changelog) for a list of all the rest of the included updates.

closed 2020-11-30 by christianhaitian