christianhaitian/arkos #42: Emulation issues after system update

dolladollabillyall, 2020-12-01:
Hello, my PS1 emulation was fine on my RG351P but then when updating to 1.4, there is now graphical stuttering and audio crackling in the games. I have the needed bios folders and all that.

Also, since before I updated, N64 emulation has been running at 5-15 FPS and changing the settings does not improve this. Can these two things be worked on in the future? Thank you so much for the awesome work you’ve done so far!

christianhaitian, 2020-12-01:
What are some example PS1 games you having this issue with?  For N64, emulation is not going to be very good right now.  You can try the standalone core as it has better performance.  See this link on how to change cores.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-change-emulator-cores-in-ArkOS

dolladollabillyall, 2020-12-01:
I’ve tested Ape Escape, Parasite Eve 2, Metal Gear Solid, and Dino Crisis 2, whatever I try has the same two issues. 

Thanks, I will check out the core for the N64.

christianhaitian, 2020-12-01:
I just tried Ape Escape and Metal Gear Solid on my end and I'm not experiencing this stuttering issue.  Try going to the retroarch menu->Options->Frameskip 0.  Maybe you have frameskip on that's causing the issue?

The standalone buttons work just fine on my end.  The link to the button mappings for mupen64plus standalone is here:  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS.  Quitting should be select+start.  Is this not what's happening for you?

dolladollabillyall, 2020-12-01:
I remember now having changed the cfg file for Mupen, I will change that back to what you have and the buttons should work.

No frameskip on PS1 enabled.

christianhaitian, 2020-12-01:
What program did you use to image the sd card with ArkOS?

dolladollabillyall, 2020-12-01:
I had used Apple Pi Baker on my Mac to do the flashing.

christianhaitian, 2020-12-01:
Any way you can try another SD card with the latest ArkOS loaded on it and see if you still have the issue there?

christianhaitian, 2020-12-01:
Hi, can you help me test something?  I had changed how the rumble works for dreamcast, N64, and psx and I wonder if that's what caused the performance issue.  Can you try the attached file that will revert those emulators to the previous versions to see if that restores the performance?  All you need to do is put the attached file in /roms/backup then go to Options/Advanced/Restore Settings.  Make sure to set N64 to the default emulator and try it and let me know what happens.  Thanks,
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5622074/arkosbackup.tar.gz)

dolladollabillyall, 2020-12-01:
Sure, after work I will test out that file. If that doesn’t work, I will try flashing another SD card to test out PS1 performance on there. I am using a Sandisk card.

migatora, 2020-12-01:
Same here: after update to 1.4. Sound in  NDS games does not work correctly and in Doom does not work at all. I had to install 1.4 from scratch them setup Doom and copy my NDS games, for them to work again correctly. Update the version via internet creates issues on games already installed on the device. It is not transparent.

dolladollabillyall, 2020-12-01:
migatora, I have the manual 1.4 file on my computer and can flash it to my SD card - does that delete all my roms? I have a backup on another SD card so I can move them if need be, it will just take a while.

migatora, 2020-12-01:
Unfortunately, that is my experience. I used the in console update option. 1.4 was installed and all seemed correctly until I tried specifically my NDS games and Doom that were running without any issues before the 1.4 update. There I found that Doom lost the capacity of using the mp3 files and the NDS sound was terrible. Fortunately, I had all my backups, so I installed the manual 1.4 version and then restored the games and save files. As you say, it took a while but it was the only way everything worked correctly. ArkOS is excellent, but I am not confident yet on the update feature.

christianhaitian, 2020-12-01:
> Unfortunately, that is my experience. I used the in console update option. 1.4 was installed and all seemed correctly until I tried specifically my NDS games and Doom that were running without any [...]

Very strange report as the 1.4 download was created from the very same update process.  Out of curiosity, did you create flash your image from a Mac as well?

migatora, 2020-12-01:
Hi,

No. It was from Windows 10. I used SD Formatter and Win32DiskImager.

Cheers.

christianhaitian, 2020-12-01:
> Same here: after update to 1.4. Sound in NDS games does not work correctly and in Doom does not work at all. I had to install 1.4 from scratch them setup Doom and copy my NDS games, for them to work [...]

Sound went out after the 1.4 update?  Th

christianhaitian, 2020-12-01:
> Hi,
> 
> No. It was from Windows 10. I used SD Formatter and Win32DiskImager.
> 
> Cheers.

What make and size is the SD card you're using for ArkOS?

migatora, 2020-12-01:
Hi again,

64GB Netac. incredible fast :-)

christianhaitian, 2020-12-01:
> Hi again,
> 
> 64GB Netac. incredible fast :-)

Ahh.  I've tried my previous distro (TheRA-NTFS) on a Netac 64GB and 128GB card and had nothing but problems with my distro on that card.  Strange unusual issues that I never had with a Sandisk or Samsung card.  Per the main page on my wiki, I only recommend cards from this link (https://www.techradar.com/news/best-sd-and-microsd-memory-cards) as those are high quality SD cards with not just good performance but are also dependable.  I suspect your issue will return eventually with the fresh install as the Netac cards are just bad for long term dependability for arm based systems like this.

dolladollabillyall, 2020-12-01:
I was using a Sandisk card - flashed on a Mac, and then updated via the console. I will try flashing again tonight and hopefully that fixes it. I should clarify it was only PS1 that was an issue for me, the other emulators were working fine

christianhaitian, 2020-12-01:
Before you do that, try the test package I attached to the earlier response.  I'm also chatting with folks on discord who may be having issues with the updated rumble cores for Dreamcast, N64 and PSX.  Based on testing, I may revert back to the old rumble core versions if it seems like those are more stable and have better performance.

dolladollabillyall, 2020-12-02:
The file didn’t work - I wound up re-flashing and PS1 works fine now

dolladollabillyall, 2020-12-02:
Interestingly, when trying to use the backup retroarch.cfg files I had saved (from before 1.4), the problem resurfaced with the PS1 emulator. When I put in the 1.4 retroarch.cfg files from the computer download, it fixed itself.

christianhaitian, 2020-12-02:
Ok.  I've gotten reports that the new rumble enabled cores are causing issues like crashing.  Which I finally experienced as well.  I will be reverting back to the previous retroarch cores for Dreamcast(Flycast-Rumble), psx(pcsx_rearmed), and N64(parallel-N64) via an update later this evening which did not experience these issues.  I'll leave this issue open for now in case you experience an issue with this emulation again within the next few days.

dolladollabillyall, 2020-12-02:
Christian, thank you so much for all your help along the way. I finally figured out what the problem was. I’ve learned that enabling rewind support in retroarch 32 is causing the graphical and audio stuttering in PS1 games. Keeping it off keeps things fine.

closed 2020-12-02 by christianhaitian