christianhaitian/arkos #45: Drastic slowdowns

thejako, 2020-12-03:
There is a problem in drastic where some times the emulator slowdown not related with the game. It could be in the title screen or in the pause without any hard 3D graphics to emulate. It usually happen when you move the right stick (pen) the speed go down to 50% and if you change from single screen to dual screen (and then back) it fix immediately. I tested it with New Super Mario and Castlevania. This doesn't seem to happen in other OS like 351Emuelec.

I'm in the las ArkOS 1.4 (12/02) with the CPU at max (1.5Ghz)

christianhaitian, 2020-12-03:
I haven’t been able to replicate this issue on my end.  The other OS don’t allow you to use 1.5Ghz.  Try the same under 1.3Ghz normal clock and see if it happens there

thejako, 2020-12-03:
I changed the speed to 1.3 GHZ and it's the same. I recorded a video for you so you can check the problem. By the way not the actual version but a previews version of 351Elec has 1.5ghz of clock speed.

christianhaitian, 2020-12-03:
Thank you.  I was able to replicate the issue but I can't seem to find the source of why that's happening just yet.

christianhaitian, 2020-12-03:
It may just be placebo effect but I noticed less crackly sound when I set screen orientation to single and screen swap to yes.  Can you test and let me know your results?

Thanks,

thejako, 2020-12-04:
I tried it and it's the same. If you need any more test just tell me ;)

christianhaitian, 2020-12-25:
Until the source for Drastic becomes available, issues like this can't be resolved unfortunately.

closed 2020-12-25 by christianhaitian

thejako, 2021-01-06:
With the new version (01/05/2021) works great!! And the problem seems to be fixed!

christianhaitian, 2021-01-06:
Thanks for coming back and reporting this.