christianhaitian/arkos #46: Korean Language Not Working

xmassound, 2020-12-04:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
When you go to Main Menu -> Advanced Settings -> Language -> KO , the interface missing Korean font and
can't read them.

### How can the issue be reproduced? (If applicable)

### What device are you using?
RG351P

### What version of ArkOS are you on?
ArkOS 1.4

### Anything else you'd like to include that may help to review this issue or feature request?
ArkOS comes with Korean Theme called Dosirak but even this one doesn't show korean. I'd like to know where I could replace
the font to the one I have.

christianhaitian, 2020-12-04:
It would be in /usr/bin/emulationstation/resources/locale/ko folder.  You'll have to use a SSH terminal client to make the changes there.  If you have a better korean language to contribute, please forward to me so I can include it.

xmassound, 2020-12-04:
I checked .po file at given location with WinSCP. They all seem fine and well translated.
The problem is when I switch to Korean on RG351P. All the fonts are broken and can't read them. I don't know if it's character set issue or just font that is being used is missing Korean. I could test replacing the font. Where would be the location for ES system font?

christianhaitian, 2020-12-04:
Hmmmm....that may be the problem.  I may have not copied the most up to date corrected ones during the build.  See here (https://github.com/christianhaitian/EmulationStation-fcamod/tree/master/resources) for the updated fonts.  Try copying those 4 .ttf files to /usr/bin/emulationstation/resources and replace the existing ones there.  See if that fixes the issue.

xmassound, 2020-12-04:
I think it will work since I only see 2 fonts currently..(without any tweak)
I'm trying to replace them but I get permission denied error code 3. Sorry I'm really bad at this stuff.. I swithched permission to 700, 777 and stuff... What to do?

christianhaitian, 2020-12-04:
use sudo

christianhaitian, 2020-12-04:
sudo cp

xmassound, 2020-12-04:
I replaced them and it still doesn't work and I see total of 6 fonts in Resources folder:

4 newly replaced fonts (the fonts you linked)
2 existing fonts (opensans_hebrew_condensed_light.ttf, opensans_hebrew_condensed_regular.ttf)

christianhaitian, 2020-12-04:
Did you restart emulationstation after replacing them?

xmassound, 2020-12-04:
Yes, I restarted 2 times, ES restart and system restart. Then, I changed to KO. Fonts are missing and can't read them...

xmassound, 2020-12-04:
I don't if this will help but all other languages worked even before I replace 4 ttf fonts..
That's including Chinese. So maybe it's not the font issue?

christianhaitian, 2020-12-04:
I'm at a loss of what the issue could be then.

xmassound, 2020-12-04:
I temporarily found the way to show Korean.

I have put Korean font into /etc/emulationstation/themes/es-theme-nes-box/_inc/fonts
and changed the name to whatever existing font name.. there were 2 fonts so I put 2 korean fonts and replaced their names to existing ones.
After I restarted, files Advanced Menu Language swiched "AUTOMATICALLY" to KO.
Now I can ready all files.

christianhaitian, 2020-12-04:
Interesting.  Thank you for sharing this.

closed 2020-12-04 by christianhaitian