christianhaitian/arkos #48: Doom 2 not showing up under ports

Blobb56, 2020-12-04:
I've made sure that I have all the correct files in the doom 2 folder in the ports folder, and yet doom 2 is not listed at all in the ports section of the ArcOS menu. I also have a correct Doom 2 sh file in the ports folder. 

Doom 1 and quake are both working perfectly. 

I'm running the latest version of ArkOS without any other noticeable problems on a Samsung Evo Select 256gb sd card.

Thanks for any help you can give me.

christianhaitian, 2020-12-04:
Sounds like the doom 2 related files are missing for some reason.  You can grab the folder from this compressed [file](https://github.com/christianhaitian/arkos/raw/main/defaultromsfolderstructure.tar) and copy the ports folder to your sd card which should resolve the issue.

Blobb56, 2020-12-05:
I did as you said, and Doom 2 still isn't showing up under ports.

Funny thing is, though, when Ioad content in retroarch, I'm able to find and load doom 2 just fine. It's almost as if ArkOS isn't recognising that there's a doom 2 folder in ports.

christianhaitian, 2020-12-05:
the Doom2.sh file is looking for /roms/ports/doom2/Doom2.wad.  It must be in that exact location and in the proper case as linux is case sensitive.  Is your wad named and cased Doom2.wad?

Blobb56, 2020-12-05:
Yes, it's called Doom2.wad. With that exact casing and everything. And the sh file is correct as well.

christianhaitian, 2020-12-05:
Is the Doom 2.sh file still located in /roms/ports?  Can you attach a screenshot of the contents of your roms/ports folder as well as your doom2 folder?

Blobb56, 2020-12-05:
[image]
[image]
[image]
[image]

christianhaitian, 2020-12-05:
Hmmm...Now post a pic of your emulationstation ports screen.

Blobb56, 2020-12-05:
[image]

Blobb56, 2020-12-05:
[image]

Blobb56, 2020-12-05:
[image]

christianhaitian, 2020-12-05:
None of the images are viewable.  The images may need to have the resolution reduced before posting.

Blobb56, 2020-12-05:
[image]

christianhaitian, 2020-12-05:
hmmm...There are some other things showing in the ES menu that are not showing content for in the folder pics you posted earlier.  What is 3-Demon and Wolfensteind 3D (Xplosin Versi)?

Blobb56, 2020-12-05:
I have no idea. Whatever they are, I didn't add them.

christianhaitian, 2020-12-05:
Have you been editing metadata at all?

christianhaitian, 2020-12-05:
What happens when you attempt to launch 3-demon?

Blobb56, 2020-12-05:
No, at least I don't think I have.

Blobb56, 2020-12-05:
3-demon launches doom for some reason. That's really wierd.

Ah well, I've got access to doom 2 from the ports menu. Thanks for all your help.

closed 2020-12-05 by Blobb56

christianhaitian, 2020-12-05:
I think something is very wrong with the folder.  It might be best for you to just remove that ports folder completely and copy the one from the attachment I linked to above.  I also noticed that you have extra files in the Doom 2 folder that are not needed.  All that's needed for doom 2 is just the doom 2 wad, nothing more.

Jooppie, 2021-01-07:
Hi,
This happens to me also. I replaced the ports folder with the original folder and it is OK. However after scraping for metadata the menu is screwed up (3D-Demon on top now which starts Doom2).
So it reproducable (latest ARK 0S 1.5  05012021)
Ron