christianhaitian/arkos #49: Scummvm not showing in menu

n9ots, 2020-12-04:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

Updated from 1.3 to 1.4 (update via menu), before that scummvm was in the menu (emulationstation), but not after.

### How can the issue be reproduced? (If applicable)

Unsure if/how to reproduce. I would assume that update via menu vs re-imaging might.

### What device are you using?

RG351P 

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
Ver 1.4 (12012020) (original ver 1.2 (11212020) with wifi updates since)

### Anything else you'd like to include that may help to review this issue or feature request?
Tried adding "menu.svm" (blank file) to all scummvm folders I could find on the system...no effect

christianhaitian, 2020-12-04:
The correct filename should be menu.scummvm.  Not sure how the menu file would have gone missing during the update as it's not touched during any online updated.  I've corrected this information in the wiki as well.  I will keep this issue open until you can confirm that changing the extension returns scummvm to the ES menu.

n9ots, 2020-12-04:
Adding that file to the scummvm rom folder worked.
I don't remember changing anything in my rom folder before of after the update, so not sure what happened. 
Thanks for the assistance 
Have fun

closed 2020-12-04 by n9ots