christianhaitian/arkos #52: Can't run GameGear games from EmulationStation

LeBiquet, 2020-12-05:
### Describe the Issue (If applicable)

When launching gamegear games from EmulationStation, the console just shows a black screen. Game doesn't run. Can't get into Retropie Menu. Can only exit back.

When running a game through retroarch itself (loading the core and the game), there is no problem.

Can't find Gamegear in the "Emulator settings", while the games are visible in emulationstation. Even the scraper worked.

### How can the issue be reproduced? (If applicable)

Try to run a gamegear game from EmulationStation.

### What device are you using?

RG351P

### What version of ArkOS are you on?

1.4

LeBiquet, 2020-12-05:
Sega Genesis works fine

LeBiquet, 2020-12-05:
es_systems.cfg looks fine

<system>
		<name>gamegear</name>
		<fullname>Sega - Game Gear</fullname>
		<path>/roms/gamegear/</path>
		<extension>.bin .BIN .gg .GG .zip .ZIP .7z .7Z</extension>
		<command>sudo perfmax; /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/genesis_plus_gx_libretro.so %ROM%; sudo perfnorm</command>
		<platform>gamegear</platform>
		<theme>gamegear</theme>
	</system>

LeBiquet, 2020-12-05:
Sound runs.

christianhaitian, 2020-12-05:
I tried 5 different game gear games on my end and they loaded up fine through emulationstation.  You won't find emulator settings in ES for game gear because it's just configured to use one emulator which is genesis_plus_gx.  When you went into retroarch to load the game there, what core did you use to load them?

christianhaitian, 2020-12-05:
What example games are you having issues loading?

LeBiquet, 2020-12-05:
Man. You're answering fast. I deleted the config override for the gamegear directory and it works fine now. Don't know what I did to break it.

I'm really sorry to have taken some of your time. 

Thanks for the amazing work!

christianhaitian, 2020-12-05:
No problem.  Enjoy!

closed 2020-12-05 by christianhaitian