christianhaitian/arkos #53: MSX 1/2: suffixes and Incompatibilities

Gordon1972, 2020-12-06:
ArkOS 1.4 (latest online update):

MSX1/2 Issues:
Suffixes that are propagated on retrogamecorps.com (**.msx1, .msx2**) are **not working** and are **non-standard**.
The standard would be .mx1/.mx2, but in ArkOS these suffixes don't work. So far only the .rom suffix works. Maybe fMSX cannot interpret 4-lettter suffixes?

Many MSX2 games don't work at all or getting blank screens/weird resolution changes with the RetroArch (64-bit) core.

christianhaitian, 2020-12-06:
Thank you for reporting this.  I will correct the extension issue with the next update.  For MSX 2, can you try the suggestion in the link below and let me know if that works?

https://retropie.org.uk/docs/MSX/#setting-default-screen-on-msx2-and-turbor-machines

Gordon1972, 2020-12-06:
I think the suggestion to type something in is not practical on the user side,
but i looked at Retrobat and Recalbox and found that they use blueMSX
as the default MSX/MSX2 Emulator, whereas ArkOS uses fMSX. 
Maybe changing the core could make a difference?
Tried different MSX2-games (50Hz/60Hz) on both systems and all work out of the box.

christianhaitian, 2020-12-06:
Bluemsx is available in ArkOS and is the default emulator for Coleco.  If you're familiar with Retroarch, can you load the games through retroarch using the bluemsx core and confirm that it truly resolves the issue?  Then I can incorporate it as the default core for msx in the next future update.

Gordon1972, 2020-12-06:
I can confirm that the games i tested with blueMSX do work, you can set blueMSX as the new standard core
for MSX1/MSX2, but... there is no button mapping for MSX1/2 with blueMSX - Someone needs to create a new mapping that fits the MSX systems.

christianhaitian, 2020-12-08:
Thank you for helping to test blueMSX.  Today's updates includes adding blueMSX as an additional and the default MSX1/2 emulator.

closed 2020-12-08 by christianhaitian