christianhaitian/arkos #54: ZX Spectrum: control mapping Issue

Gordon1972, 2020-12-06:
ArkOS 1.4 (latest online update):

ZX Spectrum: Upon starting any ZX Spectrum game i can't control anything to play a game.
Seems the control mapping is adapted to the Kempston Joystick, and although it seems correct, the controls do not work.

I have the same issue under the DOSBox core (both 64-bit).

christianhaitian, 2020-12-06:
Zx Spectrum is a computer that allows different controls depending on the game. It’s up to the end user to decide what controls are needed for the game wanting to play and configure that in the retroarch menu or in game.  As far as I know, there’s no singular best global control for zx spectrum.  If you know of one, please provide that information.

closed 2020-12-06 by christianhaitian

solidox, 2020-12-10:
Even with joystick support many games still need a keyboard to navigate to select this input option on the actual game. The select button on the rg351p should bring up the on screen keyboard to allow you to pick the appropriate joystick type at which point it should work. A wider problem is if you bind some keyboard keys and save a core override setup. Many early games do not auto detect a joystick until it is specified. 

Also take care that in the core options that there are no keybindings entered as this will override anything you select on the input menu. My tip if you have spectrum games you want to replay to to take a save state just after you select the desired joystick input.

Let me know what game you are trying to play and I am happy to check if it works.