christianhaitian/arkos #57: Can’t move in WiFi menu

kerberw, 2020-12-08:
I tried following the faq, hitting r1 after selecting WiFi from the options menu just sits on the screen.  Tried a few different ways but was just stuck.  Thanks for your hard work, any help is appreciated!

christianhaitian, 2020-12-08:
What is the make and model of your wifi stick?

magnific0, 2020-12-09:
No solution, but I had menu freezes too when I had hardware issues. In particular I was mucking around with trying solder a USB wifi dongle internally (which has 3.3V instead of 5V Vcc), causing the dongle to not work.

So @kerberw make sure the wifi device is supported.

kerberw, 2020-12-12:
@christianhaitian I got the one suggested from retro gaming corps (sorry if these are affiliate links):

I believe it is brand "pluggable", Uses the Realtek RTL8188EUS chipset according to the amazon write up.

https://retrogamecorps.com/2020/11/01/rg351p-starter-guide/#Online

https://smile.amazon.com/gp/product/B00H28H8DU/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=thedomman-20&creative=9325&linkCode=as2&creativeASIN=B00H28H8DU&linkId=685e33f3bff24c5aae2e58d86717a8fa&sa-no-redirect=1

christianhaitian, 2020-12-12:
That wifi chip should work with ArkOS.  Did you make sure that you did not have power plugged in at the same time with the wifi adapter?  Tat will cause it not to work on the RG351P.  Also be sure that you are using the latest version of ArkOS which is linked from the main github page.  It is version 1.4.

kerberw, 2020-12-12:
@christianhaitian does the port matter? Which should I use? I have 1.3 can I flash it without losing my roms? is there a good tutorial to follow for that? I am on a mac.

christianhaitian, 2020-12-12:
Which usb c port is used does not matter as along as power is not plugged in at the same time.  If you can't update online with something like a usb to ethernet adapter, you will need to copy your roms out of the partition, flash the card, then copy them back in once completed.

kerberw, 2020-12-12:
Thanks @christianhaitian I will give it a try and report back.

kerberw, 2020-12-12:
@christianhaitian can't seem to burn the 1.4 image.

Tried a reboot, redownloading the image, wiping the disk before, and I can't get it to work. I am going to try the Mega image instead of the Gdrive. Here is the applepi log

01:32:21  Initialization Info
01:32:21  - ApplePi-Baker 2.2.3 (Build 1)
01:32:23  - macOS Version 10.16 (Build 20B29) x86-64 (64 bit application)
01:32:23  - libarchive 3.3.3, by Tim Kientzle
01:32:23  - liblzma 5.2.4, by Mike Kezner et al.
01:32:23  - zlib 1.2.11, by Greg Roelofs && Mark Adler
01:32:23  - bz2lib 1.0.6, by Julian Seward
01:32:24  No HelperTool Found - Installing HelperTool
01:32:50  HelperTool Updated to 1.8.2
01:32:56  Full Disk Access First Test: FAILED
01:32:58  Full Disk Access Second Test: FAILED
01:32:59  Full Disk Access Third Test: FAILED
01:32:59  Disk not added to list - Full Disk Access FAILED
01:36:56  Full Disk Access First Test: SUCCESS
01:36:56  Drive added to list: /dev/disk2 (256 GB Mass Storage Device)
01:37:31  Restore -- START RESTORE
01:37:32  Restore Started - Restoring File to Disk
01:37:32  Destination - /dev/disk2 (256 GB Mass Storage Device)
01:37:32  Disk Presence Check - Selected Disk Found
01:37:32  Source - File: /Users/williamkerber/Downloads/ArkOS_RG351P_v1.4.7z
01:37:32  Source - Trying to determine source size
01:37:32  Source - Source is a Compressed File
01:37:32  Source - Determining size of entries stored in Archive
01:37:32  Opening Archive /Users/williamkerber/Downloads/ArkOS_RG351P_v1.4.7z
01:37:32  Source - Size of ArkOS_RG351P_v1.4.img is (7,948,206,080 bytes)
01:37:32  Destination - Verifying Disk Capacity
01:37:32  Destination - Max disk capacity is 256,068,550,656 bytes
01:37:32  Disk Size Check - Capacity of Selected Disk PASSED
01:37:32  Unmounting - Attempting Unmount of all Partition(s) of selected Disk
01:37:32  Trying to unmount /dev/disk2
01:37:32  UnMount - Waiting for unmount to complete
01:37:32  UnMount - Waiting for unmount to complete - No disk(s) unmounted yet.
01:37:33  UnMount - Waiting for unmount to complete - No disk(s) unmounted yet.
01:37:34  UnMount - Waiting for unmount to complete - No disk(s) unmounted yet.
01:37:35  Partition Unmounted - /dev/disk2s1
01:37:35  UnMount - Waiting for unmount to complete - 1 Disk unmounted.
01:37:36  Unmounting - Completed
01:37:36  Extract IMG - Extracting IMG from Compressed Archive for Resizing
01:37:36  Extract IMG - Extracting IMG from Compressed Archive (ArkOS_RG351P_v1.4.7z, entry: ArkOS_RG351P_v1.4.img)
01:37:36  Extract IMG - Opening Compressed Archive (ArkOS_RG351P_v1.4.7z)
01:37:36  Extract IMG - Extracting Entry: ArkOS_RG351P_v1.4.img
01:37:36  Extract IMG - Destination: /Users/username/tmp/applepibaker_temporary.img
01:40:57  Expanding - Disabled expanding partition, no Linux Partition found
01:40:57  Resize IMG - Resizing and Rebuilding IMG file
01:40:57  RESIZE - Attempting to RESIZE a Linux partition in the IMG file "applepibaker_temporary.img"
01:40:57  RESIZE - Find Ext2/3/4 partition and determine size and location
01:40:57  RESIZE - STATS:
01:40:57    File Size         : 7,948,206,080 Bytes (15,523,840 Sectors)
01:40:57    Linux Partition   : 7,340,032,000 Bytes (14,336,000 Sectors)
01:40:57    Starts at         : Byte 134,217,728 (Sector 262,144)
01:40:57    Linux Target Size : 255,933,284,352 Bytes
01:40:57  RESIZE - Extracting Ext2/3/4 Linux partition
01:41:07  RESIZE - Verifying and Fixing extracted Linux Partition
01:41:08  RESIZE - Expanding Linux Partition to 255,933,284,352 Bytes
01:41:14  RESIZE - Creating new IMG file and Updating Partition Table IMG file
01:42:09  RESIZE - ERROR: Read/Write Error while copying Linux partition
01:42:11  Error - Read/Write Error while copying Linux partition

kerberw, 2020-12-12:
Reinstalled ApplePi Baker....and its working again.

kerberw, 2020-12-12:
@christianhaitian You can close this I am all good now, I appreciate Ark and your help! Thank you!

closed 2020-12-12 by kerberw