christianhaitian/arkos #59: No online access after 1.4 12072020 update

SRGilbert, 2020-12-08:
Did the update this morning and now I can’t seem to access any online functions like updating cores even though Wi-Fi appears to be connected.   I just get message saying it “failed.” Tried to rerun update and that errors out as well.

christianhaitian, 2020-12-08:
Thank you for reporting.  I will have a solution available to download that will revert that part of the update.  Please standby.

SRGilbert, 2020-12-08:
> Thank you for reporting. I will have a solution available to download that will revert that part of the update. Please standby.

Dumb question, but how can apply the fix if I can’t download the update?

christianhaitian, 2020-12-08:
Here's the magic on how to restore internet connectivity:
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5661481/arkosbackup.tar.gz)

1.  Take the attached compressed file and put in your roms/backup folder.  Do not uncompress the contents.  Just place the file as is in your roms/backup folder.  If you don't have a backup folder in the roms folder, just create it.

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.  

3.  Once the restore has been completed.  Reboot the device.  

4.  It will come back on and go into Emulationstation, and reboot again in a few seconds.  

5.  Once it's done, internet services should be restored again.  

Let me know your results when you've done this

walkebe, 2020-12-08:
it appears to be fixed!  nice work and quick work too!  thank you

christianhaitian, 2020-12-08:
Thank you for confirming.  I will make this fix and the related instructions available in the releases section for others.

SRGilbert, 2020-12-08:
Confirmed working, thanks!

closed 2020-12-08 by SRGilbert

christianhaitian, 2020-12-08:
Great!

sakkyx, 2020-12-10:
for me not working...can you help me?
thanks