christianhaitian/arkos #60: Commander Genius Support

myReaper, 2020-12-08:
Hi,

do you think it would be possible to add Commander Genius support zu ArkOS?

I really like ArkOS and would really love to play Commander Keen on the RG351P.

Thank you and Best Regards
Michael

christianhaitian, 2020-12-08:
Ahh...This the game that got me into PC games.  I will definitely review how best to bring it to ArkOS.

myReaper, 2020-12-08:
There is already a version for the RG350M it seems and from what i know the guys over at 351ELEC are already working on Commander Genius support (and have it already running maybe?), just wanted to open the Ticket here as for now I prefer ArkOS and would love to play it there also :-)

Link to the RG350M Port if needed:
https://github.com/goldmojo/Commander-Genius

christianhaitian, 2020-12-08:
Yea, I saw fewt mention that.  We work together on some items.  351ELEC is a great distro that's just getting better everyday.

christianhaitian, 2020-12-15:
This has been included in today's update.  Enjoy!

christianhaitian, 2020-12-15:
See here for more information about adding registered game data files for Command Genius. 
 https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#commander-genius-commander-keen

closed 2020-12-15 by christianhaitian