christianhaitian/arkos #63: Can't update at all

crazyjoe29, 2020-12-10:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?
I have tried updating my rg351p  with the arkOS installed and now after the 12/3/2020 update I can' t get it to update at all, what do I do?

hodoring, 2020-12-10:
Me too. Even though the Wi-Fi is valid, it doesn't seem to work. Scraper can't even do it.

christianhaitian, 2020-12-10:
See this link for the fix for this issue.  https://github.com/christianhaitian/arkos/releases/tag/v1.4.12.07.2020

Geezamo, 2020-12-12:
I've done the fix, but it's still broken for me. 
Updating gives me a "There was an error with attempting this update".

christianhaitian, 2020-12-12:
Are you sure your wifi is configured?

crazyjoe29, 2020-12-12:
I  did the fix and it worked for me and I was finally able to update it. If people are still having a hard time doing the fix and it's not working still, did you make sure that the only file in the backup folder is the unextracted backup  you downloaded from  this page  because when I did that, it worked fine.plz everyone try this and try again.

closed 2020-12-12 by christianhaitian