christianhaitian/arkos #64: Psp emulation is very slow now

crazyjoe29, 2020-12-10:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?
Since the 12/3/20 update, the pssp emulation has slowed down a lot and the frame skipping isn't helping at all, plz fix this,thanx

christianhaitian, 2020-12-10:
What game(s) are you experiencing this slowdown with?

christianhaitian, 2020-12-10:
Assuming the slowdown you're referring to is with using the ppsspp standalone emulator that is the default selected emulator, can you try the attached version to see if it restores performance?

1.  Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once the restore has been completed. Reboot the device.

Try the PPSSPP emulation and report if performance is any different.  

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5674870/arkosbackup.tar.gz)

crazyjoe29, 2020-12-10:
When set to the auto function of the emu settings of the PSP emulator my games slow down  and the music is static filled esp. In the game called mystic chronicles  and many others  and every time I play that game and a few others it tells me to try frame skipping and I did and it didn't  fix it no matter what I set it to, plz help,thanx

christianhaitian, 2020-12-10:
Ok,  have you tried the new file above to see if it resolves the issue?

crazyjoe29, 2020-12-12:
Yes, and the PSP emulator is still having a few slow down problem with graphics and the sound still crackles. I.e  the mystic chronicles PSP ROM  and a few others have this problem, is there a software fix for this? Or  if not can you make one because I am very sure other people  are having this problem too. Plz help ,thanx

christianhaitian, 2020-12-12:
I'm playing this game now using the original update from 12/07/2020 and I'm not getting any slowdowns or sound crackles.  Make sure you're using the following settings on your end:

Frame Skipping = 1
Frame skipping type = Number of frames
Auto frameskip = checked
Rendering resolution = 2x PSP
Upscale level = off

christianhaitian, 2020-12-12:
I've also tried Ghouls n Ghosts and it runs great with these settings as well.

crazyjoe29, 2020-12-13:
then what format is the ROM you are using because it still is giving me a
PSP slow down problem with this game and just yesterday I got booted out of
ys seven just before I was able to get back to the first town after the
earth dragon alter thing, what am I not doing and how can this be fixed?
Plz help me,thanx if you can.

christianhaitian, 2020-12-13:
My roms are .pbp and .iso.  I don't use .cso because compressed roms can have issues since the emulator would have to uncompress the rom during play which can cause performance issues depending on the game.  Since you mentioned that you tried different settings for performance, did you try my suggested settings above?  If so, have you tried deleting the ppsspp cache files in the easyroms/psp/ppsspp/PSP/SYSTEM/CACHE folder.  If you still have issues, try a fresh image on a different SD card with a few games and see if the issue is still there.  Report back your results.

christianhaitian, 2020-12-16:
Closed due to lack of sufficient information to address root cause of issue.  Can’t replicate on

closed 2020-12-16 by christianhaitian