christianhaitian/arkos #70: French translation

olab75, 2020-12-12:
In french translation, "SCRA**PP**EUR" title should be  "SCRA**P**EUR" (only one P).
In the menu, "SCRA**PP**ER LES NOTES" (notes scraping) should be "SCRA**P**ER LES NOTES"
_"SCRAPER LES VIDEOS" (videos scraping) is OK !"_
SCRAPPER MAINTENANT -> Only one P, please ;-)
If is it hidden somewhere else, same action : only one P :-P

christianhaitian, 2020-12-12:
Can you check this and verify that are no other issues?

https://github.com/christianhaitian/EmulationStation-fcamod/raw/master/resources/locale/fr/emulationstation2.po

olab75, 2020-12-13:
I just corrected the typos (only).
One question would be about the congugation between imperative and infinitive.
Options in a menu should be infinitive (PUSH -> APPUYER) but in a tutorial it should be imperative (APPUYEZ).
(In english it is PUSH in anyway)
BTW you do a great job ! ;-)
[emulationstation2.zip](https://github.com/christianhaitian/arkos/files/5684670/emulationstation2.zip)

christianhaitian, 2020-12-13:
Thank you for checking and providing a correction.  I will include it in a future update.  Thank you for the kind words!

christianhaitian, 2020-12-15:
This has been included in today's update.

closed 2020-12-15 by christianhaitian