christianhaitian/arkos #73: After you screen scrape, it tells you to refresh game settings. There is no option in ArkOS.

woopigfoodie, 2020-12-15:
I have tried a lot of options, but if I exit emulation station and come back in to rescrape it says I have the same number of files to rescrape. 

### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-12-15:
Which online scrape are you using? screenscraper or thegamesdb

woopigfoodie, 2020-12-15:
Screenscraper. Thank you!

christianhaitian, 2020-12-15:
Is it happening with just one particular game system?  Also did you register account with screenscraper and provide your username and password into the Emulationstation menu?

woopigfoodie, 2020-12-15:
Yes I registered and input my name and pw. It scrapes great. When scraping is done on the rg351p it gives a little message at the top that says refresh your game list to see changes. I’ll try to send you a pic later. Maybe that message is erroneous. I can’t tell if it is refreshing the list on its own. If I quit out and come back in and select to just scrape what is missing it does all 4200 again.

woopigfoodie, 2020-12-16:
To follow up on this, it looks like you don't need to hit anything to refresh the games list. However, when signed in and dl through screenscraper.fr, it takes many many tries to get to zero. I started out at like 8k, then 4200, then 1500 where I am now. I just have to keep re-scraping and re-scraping until that number falls. I am not sure why that is the case; it is scraping each file individually. I'll just keep doing that until my left-to-scrape count is zero. 
By the way, is there a way to send you some cash for all the work you have put in this? Thanks.

closed 2020-12-16 by woopigfoodie

christianhaitian, 2020-12-16:
I believe you're running into a limitation of the scraper service itself.  To get better performance, you can pay screenscraper.fr for more threads.  I've also heard you can get better performance by scraping your library from a PC first then transfer the library back to the sd card with all scraped media included and usable.  Thank you for the kind words.  I may put up a donation link in the future but for now, I'm just enjoying the experience.  Take care and enjoy!

woopigfoodie, 2020-12-16:
Gotcha. That may well be the case! 
Do put in a donation link. You and Russ over at Retro Game Corps are really the bedrock programming and reliable info sources for the rg351p. It definitely made this fun tinkering process far more successful. That has value to me.