christianhaitian/arkos #74: Fresh install- EASYROMS(2) issue. When i remove sd card and put into device it does its set up then i put sd back into pc and the EASYROM folder is hidden and in Dick Genius it says its now EASYROM(2), also all 3 partitions are different formats

chrisvidkam, 2020-12-15:


christianhaitian, 2020-12-15:
All 3 partitions are of different format.  Partition 1 is a fat32 partition used by the system for loading settings.  Partition 2 is a ext4 partition for the Linux system.  Partition 3 is a exfat partition where you can load and managed your roms.  I'm not sure why the 3rd partition is hidden on your end.  If you're using a Windows 10 version 1703 or newer machine, you just need to assign a drive letter to the EASYROMS partition so you can access it again.  From the instructions in the wiki:

_FOR WINDOWS 10 1703 or newer USERS: If you don't see a drive letter named EASYROMS when you plug the SD card into a card reader, it's most likely that Windows did not automatically assign a drive letter to that partition on your SD card. This can be resolved by going to disk management (type disk management in the search bar in Windows 10 and select the first control panel app that comes up at the top as the best match), then going to the SD card with the EASYROMS partition label, then assign a drive letter to the EASYROMS partition by right clicking on the EASYROMS partition and selecting Assign Drive Letter or Change Driver letter and Path, then follow the directions from there. Once completed, the drive should show up under My Computer. You typically only need to ever do this once on the Windows machine._

chrisvidkam, 2020-12-15:
Awesome!! It worked!! Thanks 😊

chrisvidkam, 2020-12-15:
Is there a way I can add my own cover art to games?

christianhaitian, 2020-12-15:
The easiest way to do this is to scrape your existing collection using the builtin scraper then replace the resulting images with your own.  Closing this issue ticket as the original issue was resolved.  For additional assistance, you're encouraged to join the retrogamehandheld discord.  See the main wiki page for the link.

closed 2020-12-15 by christianhaitian