christianhaitian/arkos #75: Backround musik always restarts in ES main menue

OolDEMool, 2020-12-15:
Since ArkOS V1.5, every time you go up and down in the ES main menu, the background music is restarted. Normally it always ran in an endless loop, both in the main menu and in the system submenus, until you started a game.

christianhaitian, 2020-12-15:
I can't seem to replicate this issue on my end.  I tried on 2 fresh images updated to 1.5 and on my existing daily driver and my music keeps running in an endless loop without issue.

christianhaitian, 2020-12-15:
It even continues after returning from a video screensaver as well.  I don't know.

christianhaitian, 2020-12-15:
I think I found the issue you may be experiencing.  If the start menu>sound settings>only play system specific music folder is set to on, then it restarts the music while scrolling through different systems.  If you set that off, the music from the /roms/bgmusic folder continues to loop.  Not sure if this is a setting that you turned on by accident or had on before but it's not something I've supported in ArkOS yet.  it may be feasible, I just haven't focused on it.  Make sure to turn off that feature and you should hopefully be ok.

closed 2020-12-16 by christianhaitian

OolDEMool, 2020-12-16:
Oh yes that setting made the trick. Thanks for the support & good work! :)