christianhaitian/arkos #76: Multilanguage in ES doesn't work anymore

OolDEMool, 2020-12-15:
Since ArkOS V1.5 (12152020), no matter which language you select in the ES ADVANCED OPTIONS, the language remains in English.

christianhaitian, 2020-12-15:
All languages seem to work on my end except for DE and ES so those will need to be looked at and corrected.

christianhaitian, 2020-12-15:
With a fresh image, I see the issue you're experiencing.  it seems to be related to not having a timezone set which causes the ES to reboot anytime a new language is set.  I will be creating a new image later today and will force a timezone to minimize on the occurrence of that issue.

christianhaitian, 2020-12-16:
This has been addressed in the new 1.5 image available for download.

closed 2020-12-16 by christianhaitian

OolDEMool, 2020-12-16:
It's working now with the V15 (12162020) update like a charm. Thanks for the good work! :)