christianhaitian/arkos #79: Screenscaper scraping broken ?

olab75, 2020-12-15:
It does not work anymore for me... user and password are ok...

OolDEMool, 2020-12-15:
I noticed that after the update to ArkOS V1.5 no systems are preselected in the scraper (in the menu "Scrape Now" > Systems). So even before you select START. If this is your issue? Just select "SELECT ALL" in the menu "Systems" >. Very important, then deactivate "SCUMM VIRTUAL MACHINE", "SYSTEM" and "RUN COMMAND" before you go to START.

Because the last three folders should IMHO not be shown there, because no game can be scraped there. Because if you do it, it can lead to problems. But at least you get a warning, if you have selected one of these three folders. ;)

olab75, 2020-12-15:
Well my problem was a wifi trouble... no subject here, so... :-\

closed 2020-12-15 by olab75