christianhaitian/arkos #80: Issue with main menu battery indicator

OolDEMool, 2020-12-15:
The battery indicator in the main menu does not work properly. As soon as it reaches 100% in the upper menu (even if you do the test and disconnect the charging cable, it already shows 100%), you can still charge the battery for about 1 to 2 hours (orange LED) until the battery is really fully charged (green LED with the charging cable still connected). If you charge the battery to the maximum, the battery indicator hangs at 100% for an eternally long time in battery mode.  As soon as the battery indicator is at about 20-30%, the device suddenly goes off without warning, which can lead to data loss. When you turn it back on, you only get a message that the voltage of the battery is too low. This should IMHO only happen at a much lower value like the current 20-30% in ArkOS. My RG351P is brand new, so it has a brand new battery. Something seems to be wrong with the calculated maximum and minimum voltage values in ArkOS, or to fit the size of the 3500mAh battery!? Most other comparable handheld consoles only have a 2500mAh battery installed. Maybe that's why something is calculated wrong? I don't know if it is possible to show overlays (a battery warning) while playing. That would be the best of course. But in the ArkOS (ES) menu the battery indicator should be correct, so you know approximately how long you can play without connecting a powerbank or power supply. Since I'm quite familiar with Ubuntu (I've been using Linux as OS on my computers for ages) and SSH just let me know if I can test anything special. For example reading the battery values while charging and in operation etc. You would then only have to tell me how to get the corresponding battery data in your Ubuntu version.

Oh, I almost forgot to mention. I've been using ArkOS since V1.4, and I noticed the problem there, but it's still the same with the current V1.5.

Cheers

Ool de Mool

christianhaitian, 2020-12-15:
This has been an issue since ArkOS has been available for the RG351P and any other distro available except for maybe stock since it has Anbernic's own coding in it maybe.  The only source code we have for how battery calculation is done for the unit is from [here](https://github.com/lualiliu/RG351P-linux/commit/09f652ce384593fb99a009bd22e30945a1d852bd) and [here](https://github.com/lualiliu/RG351P-linux/commit/718abd7245ab07d144e03826d7b0f161ab66bed2).  It is inaccurate but it's the best we've got so far.

christianhaitian, 2020-12-15:
I'm closing this issue as this is not something that I can fix, however, if you have suggestions, please do share or join the discoord channel and share there.

closed 2020-12-15 by christianhaitian