christianhaitian/arkos #83: No gamelist.xml for PSP games on the EASYROMS partition

MetroidFusion, 2020-12-15:
Hi

I noticed that ArkOS doesn't create a gamelist.xml file (with game descriptions etc.) in the psp folder on the EASYROMS partition. Unlike the other platforms, where it creates the gamelist.xml in their respective folders. Where is the gamelist.xml for PSP games located, and would you consider moving it to the EASYROMS partition? I'm running ArkOS V1.5 (12152020). Again, thank you so much for your awesome OS! :)

christianhaitian, 2020-12-16:
gamelist.xml is generated in the psp folder on my end.  Perhaps try scraping your roms in that folder to see if that will trigger it to generate there.

MetroidFusion, 2020-12-16:
Actually, I specifically scraped the psp folder and was surprised that it didn't create the gamelist.xml in that folder. As it does for every other platform/folder I've scraped. It did create it though (I can see the game descriptions etc. in the metadata), but I don't know where it's located.

I even tried scraping the psp folder on a another sd card/fresh installation. Same result. It generates a gamelist.xml in every platform folder, excluding the psp folder.

christianhaitian, 2020-12-16:
I believe I found the issue.  There's another gameslists folder within /home/ark/.emulationstation that had some additional gamelists.  I deleted those folders then restarted emulationstation and it placed the gamelists.xml file in the /roms/psp folder as it should have been doing already.  The new ArkOS 1.5 image coming in a few hours will have this fix in place as well.

christianhaitian, 2020-12-16:
New image version is up and available.

closed 2020-12-16 by christianhaitian