christianhaitian/arkos #84: I cannot update arkos to v1.5 with wifi

Deer-Cr, 2020-12-16:
arkOS v1.4, the version in GDRIVE of wiki. Every time I want to update through wifi, it says "There was an error with attemping this update " I'm in China, but I have connect internet through VPN. I test the wifi, it can watch video on youtube. I don't know whether it's wifi's problem.Is there any way to update without wifi?Such as download file to sd card and insert to 351p.Thanks your effort to make so good OS.  By the way, can i remap my RightStick to Dpad? Then, I can control view by rightstick. ppsspps cannot read rightstick
[image]

christianhaitian, 2020-12-16:
The ability to use the right stick in ppsspp was resolved in an update.  Not sure what's wrong with the update on your end.  You can wait for the new image that will be available in a few hours and use that since connecting to github may be a problem for you for updates.

closed 2020-12-16 by christianhaitian

christianhaitian, 2020-12-16:
The new 1.5 version is now available.