christianhaitian/arkos #87: Sleep mode

itsthescizzy, 2020-12-17:
Hit or miss while attempting to get system into/out of sleep mode. problem started with 1.4 currently on 1.5. Thought it might be a hardware issue with power button, but ran 351elec all day without issue

### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-12-17:
I need more info than this to even attempt to diagnose the issue on this.  Please provide the following information.

Did you have a game loaded while attempting to get into or out of sleep?  If yes, what games on what emulator?
Did you have wifi plugged in while going into and out of sleep?
What SD card make are you using?
What program did you use to image ArkOS onto the SD card?
Are you attempting to put the system to sleep while plugged into power?

To add a little bit, I noticed that going into sleep while wifi is plugged in by not connected to a ssid will cause the system to take a few more seconds than normal to go into sleep.

itsthescizzy, 2020-12-17:
Seems to be happening w/ and with out games loaded. I’ll hit the power and sometimes nothing happens, other times the screen will go blank then immediately return the themes/game I was in. I’ll have to hit the power button several times to get into sleep mode.

Been playing Metroid Fusion GBA
Mario bros nes
NES box theme
Not plugged in or on WiFi

Win32
Samsung evo 128 gb

Sometimes I have to hit the power button 5 or 6 times to get into sleep mode

christianhaitian, 2020-12-17:
Hmm....I have a very similar setup as my daily driver and have not experienced any of these issues with ArkOS from the first release and updated to yesterday's update.  Only difference is that I use USB IT to initially load image my card.  Do you have another SD card you can image with the latest 1.5 image and see if you have the issue with that?  If not, try to backup your roms partition, image the sd card with the latest 1.5 image download, and see if the issue persist there for you.

AdamoFarano, 2020-12-17:
After the last update, I have the same issue. Sometimes I’ll hit the power and nothing happens.

christianhaitian, 2020-12-17:
I recommend you try the same as I recommended as well.

AdamoFarano, 2020-12-17:
I have reinstalled everything using the last image and the problem is gone. Thank you

itsthescizzy, 2020-12-17:
That's good to hear, I'll reinstall tonight after work

christianhaitian, 2020-12-17:
This makes me believe that parts of the update sometime fail.  I'll work on figuring out how to make then more reliable.  I may switch to packaging updates so that there is just 1 package download needed from github versus multiples.

domramsey, 2020-12-17:
Sleep mode is not working for me on 1.5 (updated from 1.4 which had the same issue). Or rather, the device will go to sleep, but pressing the power button again never wakes it. In game or in menus.  I've not tried reflashing a fresh install yet.

closed 2020-12-19 by christianhaitian