christianhaitian/arkos #89: Wifi connection problem

domramsey, 2020-12-17:
Every time I power on the RG351P I need to delete then re-add my wifi network before it will connect. Once connected it seems to work fine, but disabling/re-enabling wifi (or even removing/replacing the dongle) does not reconnect. I need to physically delete my SSID, re-add it and enter my password again before a connection is established.

christianhaitian, 2020-12-17:
Very odd problem I haven't gotten other reports on.  Given your sleep issue as well, try flashing the new 1.5 image.

domramsey, 2020-12-18:
Ok, reflashed the latest 1.5 image and still having both the same issues (wifi connection and sleep recovery).

christianhaitian, 2020-12-18:
What is your sd card?
What program did you use to flash the card?

domramsey, 2020-12-19:
Sandisk Extreme 128GB, ApplePiBaker
That said, I just booted again and the wifi worked. Very strange as I haven't changed anything and it definitely wasn't working from boot yesterday.
Still have the same issues recovering from sleep mode though.

christianhaitian, 2020-12-19:
Can you try this version and let me know if your experience is improved with this one?  I've updated the wifi drivers.  The image also includes all updates up to the date of this post.  https://mega.nz/file/PcoBlAST#cN9oVCplJRu0nzqwtowP1-aJkJpEBnYRhKFBhni28M4

domramsey, 2020-12-19:
Ok done. Wifi seems better now. At least it worked from boot straight after restoring my settings. Thanks!

christianhaitian, 2020-12-19:
And how is sleep mode working with this version?

closed 2020-12-19 by christianhaitian