christianhaitian/arkos #90: Enhancement - lzdoom OK/Cancel button swap

romadu, 2020-12-18:
Hi,
Is it possible to recompile lzdoom, changing menu.cpp so that the OK and Cancel buttons match with the rest of ArkOS? Currently B (S) is OK and A (E) is Cancel within lzdoom.

Thank you

case KEY_JOY1:
            case KEY_JOY3:
            case KEY_JOY15:
            case KEY_PAD_A:
                mkey = MKEY_Enter;
                break;

            case KEY_JOY2:
            case KEY_JOY14:
            case KEY_PAD_B:
                mkey = MKEY_Back;
                break;

Gordon1972, 2020-12-18:
I second that. Would be a nice move to more conformity.

christianhaitian, 2020-12-19:
Done.  Will be in the next update later this evening.

romadu, 2020-12-19:
Thank you very much!

romadu, 2020-12-19:
One other thing I noticed with lzdoom is that lzdoom.ini has one button
configuration for Doom (which I like) and different configurations for
Hexen/Heretic/Strife. I copied the Doom configuration for the other games
(and possibly CHex and Urban Brawler?) for consistency.

[Doom.Bindings]
1=slot 1
2=slot 2
3=slot 3
4=slot 4
5=slot 5
6=slot 6
7=slot 7
8=slot 8
9=slot 9
0=slot 0
`=toggleconsole
Joy1=invuse
Joy2=+use
Joy3=+jump
Joy4=invnext
Joy5=+speed
Joy6=+attack
Joy7=menu_main
Joy8=togglemap
Joy10=centerview
Joy9=toggle cl_run
Joy11=weapprev
Joy12=weapnext
POV1Up=+forward
POV1Right=+moveright
POV1Down=+back
POV1Left=+moveleft

[Hexen.Bindings]
5=use ArtiInvulnerability2
6=use ArtiPork
7=use ArtiTeleportOther
8=use ArtiTeleport
9=use ArtiBlastRadius
0=useflechette
T=messagemode
\=use ArtiHealth
Joy1=invuse
Joy2=+use
Joy3=+jump
Joy4=invnext
Joy5=+speed
Joy6=+attack
Joy7=menu_main
Joy8=togglemap
Joy9=toggle cl_run
Joy10=centerview
Joy11=weapprev
Joy12=weapnext
Joy14=showpop 3
Joy15=showpop 2
Joy16=showpop 1
POV1Up=+forward
POV1Right=+moveright
POV1Down=+back
POV1Left=+moveleft

[Strife.Bindings]
1=slot 1
2=slot 2
3=slot 3
4=slot 4
5=slot 5
6=slot 6
7=slot 7
8=slot 8
9=slot 9
0=slot 0
T=messagemode
`=toggleconsole
Joy1=invuse
Joy2=+use
Joy3=+jump
Joy4=invnext
Joy5=+speed
Joy6=+attack
Joy7=menu_main
Joy8=togglemap
Joy9=toggle cl_run
Joy10=centerview
Joy11=weapprev
Joy12=weapnext
Joy14=showpop 3
Joy15=showpop 2
Joy16=showpop 1
POV1Up=+forward
POV1Right=+moveright
POV1Down=+back
POV1Left=+moveleft

[Heretic.Bindings]
1=slot 1
2=slot 2
3=slot 3
4=slot 4
5=slot 5
6=slot 6
7=slot 7
8=slot 8
9=slot 9
0=slot 0
Backspace=use ArtiTomeOfPower
T=messagemode
`=toggleconsole
Joy1=invuse
Joy2=+use
Joy3=+jump
Joy5=+speed
Joy6=+attack
Joy7=menu_main
Joy8=togglemap
Joy9=toggle cl_run
Joy10=centerview
Joy11=weapprev
Joy12=weapnext
Joy14=showpop 3
Joy15=showpop 2
Joy16=showpop 1
POV1Up=+forward
POV1Right=+moveright
POV1Down=+back
POV1Left=+moveleft

christianhaitian, 2020-12-19:
The button swap of OK/Cancel is available in today's update.

closed 2020-12-19 by christianhaitian