christianhaitian/arkos #92: Feature Request: Add Menu Button to use Hexen, Heretic, Chex Quest in lzdoom

Gordon1972, 2020-12-18:
ArkOS 1.5 (latest version) -> lzdoom

I've tested lzdoom with Hexen, Heretic and Chex Quest and the games run fine, but there is no button to get
back to the game's menu, so one has to reset the RG351p.

Maybe it's possible to add a config or a button layout, so that we can enjoy these games as a bonus.

Gordon1972, 2020-12-20:
Any chance we get a config for these games before Xmas?
Just asking. ;)

christianhaitian, 2020-12-21:
I need to understand the request a little better.  Is it that once you start one of these games, there's no menu exit option like there is with Doom?

Gordon1972, 2020-12-21:
Exactly. There is no button defined to get into the games' menu. So you are not able to quit back to ArkOS or change settings in a game.

christianhaitian, 2020-12-21:
This has been accommodated in today's update.

closed 2020-12-21 by christianhaitian