christianhaitian/arkos #94: Sega Genesis

marcosgeorgopoulos, 2020-12-18:
Hi,

I’ve updated to the latest version. 1.5 ( 12182020 ). Since updating I cannot load any sega genesis games. By I am getting the error message, failed to extract content from archive. 

All other systems seem to be working still.

christianhaitian, 2020-12-18:
I just tested Alien3 (.zip), Comix Zone (.zip), Flash Point (.bin) and Fix it Felix Jr(.bin.  All loaded fine on my end on my daily driver that's updated to 1.5 (12182020) and a 1.5 image I upgraded to 12182020.  I can't replicate this issue so far.  What format are your games in?

marcosgeorgopoulos, 2020-12-19:
Thank you. They are in .zip format. It’s odd. If I backup my easyroms files and do a backup and copy that file off, then rebuilt, restore and copy my easyroms files back over what would I lose?

christianhaitian, 2020-12-19:
You should be ok to that.  For added insurance, you can take an image of your current sd card using USB IT (https://www.alexpage.de/usb-image-tool/download/) in case you lose something.  The issues you're describing leads me to believe something isn't quite right.  Make sure of the following:

You're using a good quality micro sd card such ad Sandisk or Samsung.  no name brand cards and other cheap brands like Netac will eventually lead to strange issues like what you're describing.

Make sure you're using a program like USB IT to image the card.  Balena Etcher is not recommended for imaging ArkOS as it leads to performance and usability issues like what you're describing.

marcosgeorgopoulos, 2020-12-19:
Thank you.

yeah I have a good quality card. Everything else works perfectly, all of the other systems etc. 

I didn’t use etcher. 

Cheers.

it doesn’t seem to be related to ArkOS so I will close the ticket. Cheers

closed 2020-12-19 by marcosgeorgopoulos