christianhaitian/arkos #96: Wifi Freezing issue

BepisWildherry, 2020-12-19:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Whenever i try to connect my device to wifi using the wifi dongle that was included with the rg351p or the netgear a6100 which was good on the compatibility list my entire device freezes completely requiring a full restart. as soon as i press the + sign in the add network wifi menu the device becomes inoperable. 
### How can the issue be reproduced? (If applicable)
open the wifi menu in settings and press the plus sign to add network info.

### What device are you using?
rg351p

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1.5 the latest release as of 18DEC2020

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-12-19:
What SD card are you using?
What program did you use to image the SD card?
Did the wifi dongles work without issue on the stock OS with the same usb a to usb c adapter?

BepisWildherry, 2020-12-19:
Its a sandisk extreme 128 gb
I used win 32 DiskImager
And yes I was able to insert the default sd card with the original os and connect to my wifi using the dongle without issue

christianhaitian, 2020-12-19:
Try reimaging with USB IT (https://www.alexpage.de/usb-image-tool/download/) instead and see if you get better results with that.  I use those 2 same wifi dongles without issue on my end constantly.

BepisWildherry, 2020-12-19:
ill reimage and update you once its done, thanks!

christianhaitian, 2020-12-19:
Also, make sure you not charging the device while using wifi at the same time.

BepisWildherry, 2020-12-19:
just finished reimaging and im still experiencing the same issue unfortunately

christianhaitian, 2020-12-19:
Very strange.  You do not have the unit charging at the same time while using the wifi module correct?

BepisWildherry, 2020-12-19:
No the only otg port being used is for the wifi dongle

christianhaitian, 2020-12-19:
Do you have another usb a to usb c adapter you can try?

BepisWildherry, 2020-12-19:
regrettably no i only have the one

christianhaitian, 2020-12-19:
What happens when you try to go setup your wifi without the module plugged in?  Does it still freeze?

BepisWildherry, 2020-12-19:
it does not freeze, it takes me to a second menu that is empty

christianhaitian, 2020-12-19:
Ok.  So that means it's having issues communicating with the wifi modules.  How much battery power does your unit have left?

BepisWildherry, 2020-12-19:
currently im at 60%

christianhaitian, 2020-12-19:
The battery meter is not very accurate.  Try charging the unit for about 30 minutes then try your wifi modules again and see if there's any difference.

BepisWildherry, 2020-12-19:
will do, ill keep you updated

BepisWildherry, 2020-12-19:
so i just tried again and still no luck. its at 100% charge now

christianhaitian, 2020-12-19:
Ok.  I have a test image with an updated kernel I am working on getting uploaded.  I'd like for you to test that and see if you have better success with that.  It will be a couple of hours before it's available.  My upload speed is slow.

BepisWildherry, 2020-12-19:
no problem, if i dont get it tonight ill just look in the morning. i appreciate the help!

christianhaitian, 2020-12-19:
Please try this version and let me know if your experience is improved with this one?  I've updated the wifi drivers.  The image also includes all updates up to the date of this post.  https://mega.nz/file/PcoBlAST#cN9oVCplJRu0nzqwtowP1-aJkJpEBnYRhKFBhni28M4

christianhaitian, 2020-12-21:
Any update on this?

BepisWildherry, 2020-12-23:
Sorry i have been away from my pc for the last few days. i am downloading from your mega link now

BepisWildherry, 2020-12-23:
Great! whatever you did fixed the issue! i was able to connect to my wifi using the same dongle. Thank you very much!

christianhaitian, 2020-12-23:
Good to know.  I will include these kernel module updates in the next stable image in the future.

closed 2020-12-23 by christianhaitian