christianhaitian/arkos #97: New PPSSPP Version: Massive Performance loss

Gordon1972, 2020-12-19:
ArkOS 1.5 (latest build) -> PPSSPP 1.10.3

Have tested the newly implemented PPSSPP version with Wipeout Pulse and a few other games.
Now compared to my tests i ran just yesterday with the OLDER Version, i must say there are at least 40% slowdowns
with the NEW version. Wasn't this to be supposed to be the other way round?

Only disabling Backbuffer Effects got me a few frames extra.
Maybe someone can look into this, thank you.

christianhaitian, 2020-12-19:
Try setting frameskip to 1, auto frameskip is checked, and Force Max FPS to auto.  If still not good, try forcing max fps to 30.  Also make sure Rendering Resolution is set to auto 1:1.

domramsey, 2020-12-19:
Same issue here with the latest update even with those settings, it's significantly slower for me. Also noticed an issue on Ridge Racer that I can't get the map/speed overlay to show now. Not sure if that's me having done something though!

christianhaitian, 2020-12-19:
I’m now seeing the same slowdowns.  I’ll setup an update to revert to the previous version.

christianhaitian, 2020-12-19:
The update to revert ppsspp back to the previous version is ready.  Thanks for quickly reporting the issue with the new ppsspp.

closed 2020-12-19 by christianhaitian