christianhaitian/arkos #98: Screenshot failed retro achievement

AlexKCN, 2020-12-19:
When you take a screenshot it locks and returns to the menu, and in the retro achievement does not save

christianhaitian, 2020-12-19:
This seems to be related to notifications being on for it.  In Retroarch, go to settings>On-Screen Display>On-Screen Notifications>Notification Visibility>Screenshot Notifications>off.  This seems to fix the issue.

AlexKCN, 2020-12-19:
[image]

For manualy screenshot it you spoke work, but

 Still fail to save when you got a achivement

AlexKCN, 2020-12-19:
Fail the automatic screenshot on achievements

christianhaitian, 2020-12-20:
Hmmm...The only way it would take the automatic screenshot is when I set it to save to the /home/ark/.config/retroarch/screenshots folder.  Settings>Directory>Screenshot.  Not sure why it won't save that to the content directory.  Maybe an incompatibility with exfat

AlexKCN, 2020-12-20:
Ok, I will try, thanks

AlexKCN, 2020-12-20:
It's works!!! Very thanks bro. 🥇

closed 2020-12-20 by christianhaitian